FreeCAD/FreeCAD-library #115: Sliding 202 panes does not show in FreeCAD 0.15

nicoguaro, 2015-09-23:
I tried to open the [piece](https://github.com/FreeCAD/FreeCAD-library/blob/master/Architectural%20Parts/Windows/Sliding%202-panes%20window.fcstd) to generate an STL from it, but it does not show. When I try opening the STEP it works.

**This is the info of my system:**

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

yorikvanhavre, 2015-09-23:
Hm this file seems to have serious problems indeed... I think it'll be faster to redo it from scratch

yorikvanhavre, 2015-09-25:
Ok this is fixed now (I added the stl file too)

closed 2015-09-25 by yorikvanhavre