FreeCAD/FreeCAD-library #137: 0.16 on developing: PartsLibrary.FCMacro does not work

Almisuifre, 2015-12-11:
Description : Nothing displays
------------------------------------------------------------------------------------------
Steps To Reproduce : 
- I import my fork locally (OK)
- I slipped into the .FreeCAD Library.FCMacro Share file (OK)
- I opened FreeCAD then I launched the macro (OK)
- A dialog (non-modal) open (OK)
- I select my local path that is well within the database (OK)
- And he goes nothing!
------------------------------------------------------------------------------------------
[image]
[image]

Additional Information : 
Some error messages in the report view :

python-git not found. Git-related functions are disabled
Your library is not a valid Git repository. Please clone it with git first.

Some messages in the python console :

Python 2.7.6 (default, Jun 22 2015, 18:01:27) 
[GCC 4.8.2] on linux2
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>>
[image]

grey3050, 2015-12-14:
I'm having the same issue in windows 10,  freecad 15 with the normal stable freecad release.

yorikvanhavre, 2015-12-15:
Could any of you try this?
1) rename PartsLibrary.FCMacro (the one that you have copied to your macros folder) to PartsLibrary.py (change .FCMacro -> .py)
2) In the freecad console: `import PartsLibrary`
And tell me if any message error appears?
Thanks

Almisuifre, 2015-12-15:
For me, nothing has changed...

python-git not found. Git-related functions are disabled
Your library is not a valid Git repository. Please clone it with git first.

But I understand the process ... The window remains empty dan sles both.

yorikvanhavre, 2015-12-15:
Ok I think I found the problem... I'm committing a new version of the macro.

closed 2015-12-15 by yorikvanhavre

grey3050, 2015-12-15:
Yorik,

The new macro apparently has a variable "updatebutton" that needs to be defined before it will run.  I don't know python or I'd try and fix it :)

New error as follows after update to your fixed version:
-------
python-git not found. Git-related functions are disabled
Traceback (most recent call last):
  File "C:/Users/grey/AppData/Roaming/FreeCAD/PartsLibrary.FCMacro", line 385, in <module>
    m.addDockWidget(QtCore.Qt.RightDockWidgetArea,ExpDockWidget(LIBRARYPATH))
  File "C:/Users/grey/AppData/Roaming/FreeCAD/PartsLibrary.FCMacro", line 203, in __init__
    updatebutton.setEnabled(False)
<type 'exceptions.NameError'>: global name 'updatebutton' is not defined

Date: Tue, 15 Dec 2015 08:05:15 -0800

grey3050, 2015-12-15:
I think broke a define when you fixed it, updatebutton is now not defined.

python-git not found. Git-related functions are disabled
Traceback (most recent call last):
  File "C:/Users/grey/AppData/Roaming/FreeCAD/PartsLibrary.FCMacro", line 385, in <module>
    m.addDockWidget(QtCore.Qt.RightDockWidgetArea,ExpDockWidget(LIBRARYPATH))
  File "C:/Users/grey/AppData/Roaming/FreeCAD/PartsLibrary.FCMacro", line 203, in __init__
    updatebutton.setEnabled(False)
<type 'exceptions.NameError'>: global name 'updatebutton' is not defined

yorikvanhavre, 2015-12-15:
Doh.. stupid mistake. Should be fixed now.

grey3050, 2015-12-15:
works perfectly now under windows 10, freecad 15, thanks for you quick responses!

Date: Tue, 15 Dec 2015 10:11:34 -0800

Almisuifre, 2015-12-16:
Hello,

I just changed the script ... And, it works :)

Thank you so much for changing the script.

commit 40d68275 (2015-12-15): Allow the macro to work without git - fixes #137