FreeCAD/FreeCAD-library #144: Sol Expert

galou, 2016-02-04:
Are we sure we want to have the "Sol Expert" folder in this repository? It doesn't appear to me to be close to any standard.

If we really want it, all spaces should be removed from the names and all files should be in English.

Almisuifre, 2016-02-04:
I designed this library in relation to specific components for vehicles in 1:87 scale. One can get these components on the website Sol Expert Group ...

Several solutions possible.
1 / If this is concidered as unnecessary, remove it but remain in my github github perso ... My nefera more copy of the official github ...
2 / It is misplaced, then move, it poses any worries me.
3 / That remains as it allows and to benefit many

galou, 2016-02-04:
Is 1:87 a standard modeling scale? In this case, we should only rename the directory to something more neutral like "Modeling/1_87" and let "Sol Expert" as subdirectory.

wood-galaxy, 2016-02-04:
The models in the library does not have to be standard. However I agree that it might be to create a category named like "Reduced Models" ?

normandc, 2016-02-05:
I believe the English expression is "scale model". https://en.wikipedia.org/wiki/Scale_model

So the directory tree could be FreeCAD-library/Scale Models/Sol Expert/1_87.

Almisuifre, 2016-02-06:
It would be perfect if it could return to this logic.

wood-galaxy, 2016-02-06:
A directory have been create. However about file name, I don't think english was so important, I prefer having more object than forcing people to translatefile name. One could always change the name and provide a pull request.

closed 2016-02-06 by wood-galaxy