FreeCAD/FreeCAD-library #188: Each directory could have a README.md that shows screenshots

luzpaz, 2017-02-16:
...this would help folks browse the repo instead of needing to download files to view their contents.

Obijuan, 2017-02-16:
I think it is a great idea!

luzpaz, 2017-02-16:
@Obijuan thanks :)
Perhaps you can make it a guideline that it become part of the contribution workflow to make a png of their submission and at it as a README.md to said directory ?

luzpaz, 2017-02-17:
Ideas for submissions guidelines:
* must be submitted with a README.md file 
* README.md template (needs to be decided on)
* image format (png?) must be certain size 
* image name must not have white space in it.

infthi, 2017-04-24:
FIY github has a built-in viewer for stl files

battlesnake, 2019-05-06:
Could possibly have a script hooked to the repo the generates several views of each part and publishes them to a separate repo/site/archive rather than putting them in the source repo, to prevent the size from ballooning too rapidly

yorikvanhavre, 2019-05-08:
That might be a good idea, but it would require a server to run this script, etc...
At the moment the library weights more or less 1Gb, including:
- The .git folder (not present if you dont use git): 315Mb
- jpg files: 512Kb
- png files: 1.7Mb
- step files: 342Mb
- stl files: 164Mb
- brep files: 161Mb
- FCStd files: 143Mb

So the image files really are very small, not even 1% of the total size. Also, FreeCAD has the ability to save a thumbnail inside the file, which would make these image files not necessary...

What we should do, maybe, is try to get rid of all the brep and step files. For that, we should better the tools in FreeCAD to offer several ways to import an FCStd file, to make these additional formats unnecessary (basically the ability to import them as one object). Then, make sure every brep or step file here has a corresponding FCStd file.

silopolis, 2019-05-10:
Why sacrifice STEP and save STL?
AFAIC, if I had to delete one, it'd be STL 🤔

easyw, 2019-05-10:
@yorikvanhavre 
we could remove brep format (which adds nothing compared to .FCStd file) ...
STEP files are a bit depending on what release of FC is used to export and which options are selected... so I would keep this format in the library... eventually they can be added as zip or even as '.stpZ' format (which is a zip std step format that can be read and saved in FC).
The file size will drop 10 times or more normally...

yorikvanhavre, 2019-05-13:
stl is useful because you can visualize them in 3D in github (who knows one day they support FCStd files :smile: )
Probably the best way to proceed with all this is not to have the parts library as a simple FreeCAD addon anymore, but a separate system where you can download parts on-the-fly, as you need, and not simply download the whole thing as a bundle... But that requires basically coding a whole new system for it. If anyone is interested in doing that, I'm interested in helping ;)

battlesnake, 2019-05-13:
I could possibly take that on, but not for a few months due to current paid
projects.

We could also look at the possibility of forking an existing
package-management system e.g. CTAN/CPAN/CRAN/pacman/etc, to minimise the
amount of actual code that needs to be written.

yorikvanhavre, 2019-05-13:
> I could possibly take that on, but not for a few months due to current paid projects.

That would be great!!

This can be pretty simple... A dialog that basically shows a tree structure with files, and on double-clicking any of them it would download it and insert it in FreeCAD. There is a working example of this in https://github.com/yorikvanhavre/BIM_Workbench/blob/master/BimLibrary.py but the way the tree is populated there is automatic, from the filesystem. Here we'd need to do it manually.

The most annoying thing would be to get that files & directory structure from github. But maybe there is some cool API functionality to get that...

Cool too would be that such a tool wouldn't prevent using the library as we do now (clone the whole thing locally). It would just be an alternative, online UI for it...

battlesnake, 2019-05-13:
I'm currently developing similar things for synchronising data with
satellites, so hopefully if I have time later in the year then I'll have a
decent idea of how it should all work!

Mambix, 2025-09-09:
Can I highjack this tread for #444 

Can also add assembly files to show how things fit together where it makes sense.