FreeCAD/FreeCAD-library #190: Windows 10: problem during addons installation and unpacking file FreeCAD-library-master.zip.

mbi33, 2017-02-17:
# Win10, FreeCAD 0.16, parts-library: 
### Small problems with parts-library

1. Macro "addons_installer.FCMacro" **hangs** during installation parts-library. Hangs well during the removal of parts-library too. That's why the folder "*C: \ Users \ xxx \ AppData \ Roaming \ FreeCAD \ Mod*" contains **incomplete data**.
2. When I used file FreeCAD-library-master.zip. WinRAR **interrupts** unpacking the zip file on the wrong file names.

## Cause:

-  **colon character** in the file name in the Win10 unacceptable, such as "*FreeCAD-library-master \ Mechanical Parts \ 4-wire thread \ Captura de pantalla de 2016-12-31 13:12:02.png*" (4 pieces)
- **Illegal characters** in "*FreeCAD-library-master \ Mechanical Parts \ Sigma profiles \ .*" files **Ayarxxx180.stp**, **Profilxxx45x45.fcstd**.
 
Please correct it.
mbi33

Mark196, 2018-03-12:
Similar problem.
Widows 10 64 bit Pro
Freecad OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6712 (Git)
Build type: Release
Branch: releases/FreeCAD-0-16
Hash: da2d364457257a7a8c6fb2137cea12c45becd71a
Python version: 2.7.8
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

parts-library hangs on removal and installation. If you close the add-on installer and restart the installer marks it as installed but it is not available. Hangs on removal and updating. I've tried reinstalling it but to no effect.

Mark196

sgrogan, 2018-03-12:
see here: https://forum.freecadweb.org/viewtopic.php?t=27189
I'll do additional testing on Win

hyarion, 2020-12-31:
This issue was fixed in https://github.com/FreeCAD/FreeCAD-library/pull/198 by @Obijuan. The file no longer has a colon character in the name so I think the issue can be closed now.

closed 2021-03-07 by normandc