FreeCAD/FreeCAD-library #196: Unable to checkout working tree on Windows

chrisjaquet, 2017-05-17:
When cloning the repo on windows, I get the following error:
```
POST git-upload-pack (151 bytes)
remote: Total 9299 (delta 0), reused 0 (delta 0), pack-reused 9299        
unable to create file Mechanical Parts/4-wire thread/Captura de pantalla de 2016-12-31 13:12:02.png: Invalid argument
unable to create file Mechanical Parts/4-wire thread/Captura de pantalla de 2016-12-31 14:13:15.png: Invalid argument
unable to create file Mechanical Parts/4-wire thread/Captura de pantalla de 2016-12-31 14:14:07.png: Invalid argument
unable to create file Mechanical Parts/4-wire thread/Captura de pantalla de 2016-12-31 14:18:17.png: Invalid argument
unable to checkout working tree
Clone succeeded, but checkout failed.
```

This is due to the invalid ':' character appearing in the indicated filenames caused by merging in pull request 181 (commit 497628f37e9551f6715354c9d013d7861c0880b4) on 2016-12-31 18:47:33.

Windows users will be unable to checkout until the files have been renamed. Thanks in advance.

rautamiekka, 2017-05-17:
**My 2 cents:** when I get a character that's illegal on Window$ and I have to care about it, I replace it with an underscore: 1*`_` for every illegal character. It may look rather ugly, but at least the number of characters is the same and the character is accepted in most systems (except for those that use it for the same purpose as NIX uses `/`), so it's also programmatically predictable.

luzpaz, 2017-05-17:
Either way those image files should be renamed

Obijuan, 2017-06-14:
It should be fixed now. Please try again. Thanks :-)

closed 2017-06-14 by Obijuan

chrisjaquet, 2017-06-14:
Thanks - I am able to checkout again :-) 

I appreciate the effort.

commit 9b622f4c (2017-06-14): 4-wire thread: screenshots renamed. It fixes #196