FreeCAD/FreeCAD-library #200: Qt5 compatibility issue

kkremitzki, 2017-07-13:
Hi, I'm running into the following error using this on Arch Linux's FreeCAD-Qt5 build:
```
Traceback (most recent call last):
  File "/home/kurt/.FreeCAD/Macro/PartsLibrary.FCMacro", line 449, in <module>
    m.addDockWidget(QtCore.Qt.RightDockWidgetArea,ExpDockWidget(LIBRARYPATH))
  File "/home/kurt/.FreeCAD/Macro/PartsLibrary.FCMacro", line 215, in __init__
    self.retranslateUi()
  File "/home/kurt/.FreeCAD/Macro/PartsLibrary.FCMacro", line 220, in retranslateUi
    self.setWindowTitle(QtGui.QApplication.translate("PartsLibrary", "Parts Library", None, QtGui.QApplication.UnicodeUTF8))
<type 'exceptions.AttributeError'>: type object 'PySide2.QtWidgets.QApplication' has no attribute 'UnicodeUTF8'
```

Version info:
```OS: Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.11528 (Git)
Build type: Release
Branch: master
Hash: d31be3117de386b4f6cd3444e0b330db56dc6ba2
Python version: 2.7.13
Qt version: 5.9.1
Coin version: 3.1.3
OCC version: 7.1.0
```

peterlama, 2017-07-13:
Encoding for translate is not specified in Qt5 because it always uses utf8 as mentioned here: https://wiki.qt.io/Transition_from_Qt_4.x_to_Qt5. A try/except will have to be added. For example, https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Draft/DraftGui.py#L51

normandc, 2019-03-04:
Guys, this is still not fixed, Qt5 builds are becoming commonplace and the PartsLibrary.FCMacro can no longer be launched...

namezis, 2019-04-28:
I have installed FreeCAD 0.18, and the Qt5 does not tolerate the "QtGui.QApplication.UnicodeUTF8".

peterlama is correct.

Following direction from https://wiki.qt.io/Transition_from_Qt_4.x_to_Qt5 and removing all encoding parameters fixed the issue. After changes the script is still functional in FreeCAD 0.18.

I have attached it here: [PartsLibrary.FCMacro.txt](https://github.com/FreeCAD/FreeCAD-library/files/3124732/PartsLibrary.FCMacro.txt)

hyarion, 2020-12-31:
As far as I can see, this issue was fixed in the PR https://github.com/FreeCAD/FreeCAD-library/pull/252.
@kkremitzki can you verify it and close this issue if it is

closed 2021-02-11 by kkremitzki