FreeCAD/FreeCAD-library #205: 438, 5, 'print "Library path ", LIBRARYPATH, "not found."\n'))

LaurentRouchairoles, 2017-08-02:
I have a message : <unknown exception traceback><type 'exceptions.IndentationError'>: ('expected an indented block', ('/home/lolo/.FreeCAD/Macro/PartsLibrary.FCMacro', 438, 5, 'print "Library path ", LIBRARYPATH, "not found."\n')) 

[PartsLibrary.FCMacro.tar.gz](https://github.com/FreeCAD/FreeCAD-library/files/1194646/PartsLibrary.FCMacro.tar.gz)

The file is ./freecad/Macro I launch freecad but when I select the macro and select launch I have that message : <unknown exception traceback><type 'exceptions.IndentationError'>: ('expected an indented block', ('/home/lolo/.FreeCAD/Macro/PartsLibrary.FCMacro', 438, 5, 'print "Library path ", LIBRARYPATH, "not found."\n'))

hyarion, 2020-12-31:
The issue seems to have been a missing indentation on the last line in the PartsLibrary.FCMacro file.
The strange thing is that the file never had that issue, I've checked the history of that particular line and it has been correctly indented for at least 7 years as far as I can tell.

My conclusion is that this must have been a user error. So for that reason, I think it is safe to close this issue.

normandc, 2021-03-07:
It's more than 3 years old anyway, using the now prehistoric V0.16 version. V0.19 is due for release as soon as packagers resolve library issues, @LaurentRouchairoles please open a new issue if you still encounter the issue.

closed 2021-03-07 by normandc