FreeCAD/FreeCAD-library #208: Cloud it be a bicycle store ?

FrViPofm, 2017-08-14:
I have started one :  https://github.com/FrViPofm/Bike3D

But it would rather be among the other stores...

normandc, 2019-01-10:
Please explain the purpose of this issue. The issue title and description make no sense to me.

PrzemoF, 2019-01-10:
It seems strange, but .... I read it as : please move FreeCAD repo to gitlab. I'd close it if I could.

P.S.  I just learned github was acquired by MS. - get ready for embrace-extend-extinguish?

dlastov, 2019-02-24:
If you want to contribute with bicycle models to the FreeCAD-library, so the instructions are here:
https://github.com/FreeCAD/FreeCAD-library#sharing-your-models-from-the-macro

normandc, 2021-03-07:
Closing this issue. As @dlastov mentioned, a pull request is probably better suited - assuming @FrViPofm is still willing to have their work hosted on Github.

closed 2021-03-07 by normandc