FreeCAD/FreeCAD-library #233: FreeCAD 0.17 with Win 10 - FreeCAD-library - PartsLibraryFC.Macro connection crashes app & gives Errors.

V8Skater, 2018-11-26:
System:
Windows 10 upgrades from 7
64 bit O/S

luzpaz, 2018-11-26:
I recently saw a post about this in the FC forum. Can't find it ATM. Please open a forum thread to discuss this instead of an issue on this repo. Forum is the best place to get help.

luzpaz, 2018-12-03:
@kprasadvnsi does this need to be added to the original macro to be more 'future friendly' ?

kprasadvnsi, 2018-12-03:
i don't know. may be the repo admin can answer it better. i had the same problem on my linux machine and doing that fix the problem.

luzpaz, 2018-12-03:
@V8Skater please test and close issue if fix works.

V8Skater, 2018-12-05:
My parameters have already been set as kprasadvnsi described in the module, the sys already exists there. I have not changed any of them. 

I've tried to install the parts_library through the installer, however, this takes longer than I thought, and I've tried downloading the zip file through github and opening the files there. However, I'm unsure how to link these files directly to FreeCAD as the instructions are not very clear on how to do this. I have tried to create new macro and Macro folders in the bin folder, along with creating a new libs folder, though FreeCAD does not recognise these changes.

I've also tried following yorikvanhavre's advice by adding the missing import sys in the macro, However, this changes nothing. I'm aware part_library is quite a large file in total and I've waited for copious amounts of hours installing the files through the installer and github, but the instructions on github  aren't very clear. I'm not highly educated in the computer field but I can follow instructions if they're set correctly.

normandc, 2021-03-07:
@V8Skater this issue is now more than 2 years old, and FreeCAD 0.17 is no longer current. V0.19 is due for release very soon, once the packagers address some issues. Please try again with the new release when it's available and report back. Thanks.

closed 2021-12-13 by luzpaz