FreeCAD/FreeCAD-library #244: 3D rotations of .STEP file

sabrinamehlal, 2019-07-19:
I would like to get the rotation on the X,Y,Z axis and the position of a .STEP object on FreeCAD, and also get screenshots in every position of the object with Python Scripts of FreeCAD.

I saw that we can do it manually (I'm a real beginner on FreeCAD), but it's not what I want.

So I tried this, but it's doesn't seems to be the good solution :

```
>>> for i in range(360):
...     for j in range(360):
...         for k in range(360):
...             FreedCAD.getDocument("mySTEP").getObject("Part_Feature").Placement = App.Placement(App.Vector(0,0,0),App.Rotation(App.Vector(i,j,k),1))

```

As I said, I want to get screenshots of the object in all its rotations on X,Y and Z and also its position.

Is anyone can help me please?

Thank you

luzpaz, 2019-07-19:
This is not the appropriate place to make this request, right? Because it's a library of models. You'll have better exposure and more relevance if you post to the FreeCAD forum. Specifically the Python and macros subforum. Please do so and close this ticket. Thank you

normandc, 2021-03-07:
I doubt regular users can close tickets. I'd think only this repo's managers can. Doing so.

closed 2021-03-07 by normandc