FreeCAD/FreeCAD-library #251: Library doen't start. 'PySide.QtCore.QCoreApplication.translate' called with wrong argument types

rafaelrs, 2019-09-19:
Installed thru Addon Manager. Entered Macro -> Macros... -> PartsLibrary.FSMacro, then Execute and got this:
```
python-git not found. Git-related functions will be disabled
Traceback (most recent call last):
  File "/home/rafaelrs/.FreeCAD/Macro/PartsLibrary.FCMacro", line 465, in <module>
    m.addDockWidget(QtCore.Qt.RightDockWidgetArea,ExpDockWidget(LIBRARYPATH))
  File "/home/rafaelrs/.FreeCAD/Macro/PartsLibrary.FCMacro", line 231, in __init__
    self.retranslateUi()
  File "/home/rafaelrs/.FreeCAD/Macro/PartsLibrary.FCMacro", line 236, in retranslateUi
    self.setWindowTitle(translate("PartsLibrary", "Parts Library"))
  File "/home/rafaelrs/.FreeCAD/Macro/PartsLibrary.FCMacro", line 104, in translate
    return QtGui.QApplication.translate(context, text, None, utf8_decode & _encoding)
<class 'TypeError'>: 'PySide.QtCore.QCoreApplication.translate' called with wrong argument types:
  PySide.QtCore.QCoreApplication.translate(str, str, NoneType, int)
Supported signatures:
  PySide.QtCore.QCoreApplication.translate(unicode, unicode, unicode = None, PySide.QtCore.QCoreApplication.Encoding = CodecForTr)
  PySide.QtCore.QCoreApplication.translate(unicode, unicode, unicode, PySide.QtCore.QCoreApplication.Encoding, int)
```
I am using Fedora 29 with following:
```
$ uname -r
5.2.11-100.fc29.x86_64
$ rpm -qa | grep -i freecad
freecad-data-0.18.3-1.fc29.noarch
freecad-0.18.3-1.fc29.x86_64
$ pip3 list
Package             Version       
------------------- --------------
appdirs             1.4.3         
argcomplete         1.9.5         
argh                0.26.1        
asn1crypto          0.24.0        
Beaker              1.10.0        
beautifulsoup4      4.6.3         
binwalk             2.1.1         
blivet              3.1.2         
blivet-gui          2.1.11        
cairocffi           0.9.0         
CairoSVG            1.0.20        
cffi                1.11.5        
chardet             3.0.4         
configobj           5.0.6         
coverage            4.5.1         
cryptography        2.3           
cssselect           0.9.2         
cssutils            1.0.2         
cupshelpers         1.0           
cycler              0.10.0        
decorator           4.3.0         
deepdiff            3.3.0         
distro              1.3.0         
docutils            0.14          
entrypoints         0.2.3         
fros                1.1           
future              0.16.0        
gajim               1.1.3         
gpg                 1.12.0        
html5lib            1.0.1         
humanize            0.5.1         
idna                2.7           
iniparse            0.4           
initial-setup       0.3.62        
iotop               0.6           
IPy                 0.81          
isc                 2.0           
javapackages        5.3.0         
jsonpickle          0.9.4         
kerberos            1.3.0         
keyring             13.2.1        
kiwisolver          1.1.0         
langtable           0.0.41        
libvirt-python      4.7.0         
lorax               devel         
lxml                4.2.5         
Mako                1.0.12.dev0   
MarkupSafe          1.0           
matplotlib          2.2.4         
meson               0.51.1        
nagstamon           3.2.1.fedora29
nbxmpp              0.6.10        
notify2             0.3.1         
ntplib              0.3.3         
numpy               1.15.1        
olefile             0.46          
ordered-set         2.0.2         
Paste               2.0.3         
perf                0.1           
pid                 2.2.3         
Pillow              5.3.0         
pip                 18.1          
Pivy                0.6.5a0       
ply                 3.9           
precis-i18n         1.0.0         
Printrun            2.0.0rc5      
productmd           1.21          
psutil              5.4.3         
pwquality           1.4.0         
pyasn1              0.3.7         
pycairo             1.17.1        
pycollada           0.4           
pycparser           2.14          
pycrypto            2.6.1         
pycups              1.9.72        
pycurl              7.43.0.2      
pydbus              0.6.0         
pyglet              1.3.2         
PyGObject           3.30.5        
pyinotify           0.9.6         
PyIscsi             1.0           
pykickstart         3.16          
pyOpenSSL           19.0.0        
pyparsing           2.2.0         
pyparted            3.11.0        
PyQt5               5.11.3        
PyQt5-sip           4.19.13       
pyserial            3.1.1         
PySide              1.2.4         
PySocks             1.6.8         
pystray             0.14.3        
python-augeas       0.5.0         
python-dateutil     2.7.5         
python-dmidecode    3.12.2        
python-git          2018.2.1      
python-gnupg        0.3.8         
python-linux-procfs 0.6.1         
python-meh          0.47          
python-xlib         0.23          
python3-lxc         3.0.4         
python3-simplepam   0.1.5         
pytoml              0.1.18        
pytz                2018.5        
pyudev              0.21.0        
PyXB                1.2.6         
pyxdg               0.26          
PyYAML              5.1           
registries          0.1           
requests            2.20.0        
requests-file       1.4.3         
requests-ftp        0.3.1         
requests-kerberos   0.12.0        
rpm                 4.14.2.1      
rpmconf             1.0.19        
schedutils          0.6           
SecretStorage       2.3.1         
Send2Trash          1.5.0         
sepolicy            1.1           
setools             4.1.1         
setroubleshoot      1.1           
setuptools          40.8.0        
sh                  1.12.14       
simplejson          3.16.0        
simpleline          1.4           
sip                 4.19.13       
six                 1.11.0        
slip                0.6.4         
slip.dbus           0.6.4         
sos                 3.7           
SSSDConfig          2.2.0         
systemd-python      234           
tabulate            0.8.3         
Tempita             0.5.1         
torbrowser-launcher 0.3.1         
urllib3             1.24.3        
webencodings        0.5.1         
wrapt               1.10.11       
wxPython            4.0.1         
youtube-dl          2019.7.30
```

hyarion, 2020-12-31:
This has been marked as fixed by @lrochfort and by looking at the stack trace it indeed seems like it should have solved the issue.

I think this issue can be closed.

normandc, 2021-03-07:
As per @hyarion's comment above, closing this issue. FreeCAD 0.19 is due very soon, @rafaelrs please test again once it's available (hopefully in the next week or two) and open a new issue if you still encounter problems. Thanks.

closed 2021-03-07 by normandc