FreeCAD/FreeCAD-library #289: Addon Name is wrong library instead of parts_library

claus007, 2020-11-30:
it is either wrong in your description or in the list...

btw. how do I get rid of the old one ?

claus007, 2020-11-30:
[image]

normandc, 2021-03-07:
Do you still have this issue? In my version of FreeCAD, the Add-on Manager correctly lists it as parts_library.

I would suggest you check into your Macro folder, maybe you have a duplicate of the parts_library.FCMacro macro renamed "library.FCMacro". The Macro dialogue will show where your Macro folder is located. https://wiki.freecadweb.org/Macros

luzpaz, 2021-03-07:
@claus007 
1. open you default FC directory (obtain your user's default FreeCAD folder by typing the following in to the FreeCAD Python console: `FreeCAD.ConfigGet("UserAppData")` on Linux it's `~/.FreeCAD` )
2. find the `Mod` subdirectory and open it
3. look for a folder called `library` and remove it

closed 2021-03-07 by luzpaz

claus007, 2021-06-15:
it's ok by now
[image]