FreeCAD/FreeCAD-library #295: Licence(s)

Thorbijoern, 2021-01-23:
Hi,

i'm a beginner and starting to do some open (source) hardware projects with freecad.
Libraries like this make designing much faster and less tedious, but i want to give good credit to every person whose tools or parts i used to aid me.
So i'm concerned about the Licence of this Workbench. I could not find Licence information in this repo or the wiki neither about the code nor about the library of parts.
I think its important that this Workbench gets propper licences and some explanaition for the users of this workbench, so nobody accidentaly does not credit the work of others and nobody gets into legal trouble.

The People maintaining the BOLTS library made a nice writeup detailing the problems for the maintainers and the users which might aid this library too: https://www.bolts-library.org/en/docs/0.3/document/general/licensing.html

normandc, 2021-03-07:
There is a specific License section in this repo's ReadMe:

> All Parts in this repository are licensed under CC-BY 3.0 http://creativecommons.org/licenses/by/3.0/ Each Part is copyrighted by and should be attributed to its respective author(s). See commit details to find the authors of each Part.
> 
> If you are uploading parts to this repository, please make sure you are the author of the model, or otherwise that you have right to share it here under the CC-BY 3.0 license, and make sure the author is mentioned in the commit message.

Isn't this sufficient? I understand it may not be easy to find out an author through commits. I've just today noticed that a file I myself uploaded 8 years ago was reuploaded last year by another author, with inexplicable (and needless) changes to it.

Enforcing this would be extremely difficult. It would be easier if contributors populated the Project info fields directly within FreeCAD (File --> Project information menu). I admit to not having done it myself.

An easy way out would be to use a more permissive CC license without attribution, maybe? If anyone is unwilling to comply to this license, then they could just avoid uploading files.

s-light, 2021-06-13:
i think an CC0 would be a one option -
as crediting all the authors can get very tedious....
Kicad has done it in another way:
https://www.kicad.org/libraries/license/
the library as itself is CC BY SA.
but has an exception for the inclusion of parts in own projects.

hasecilu, 2022-02-25:
**Should the license be updated to CC-BY-4.0?**

In the main README.md file is stated that all parts are under the [CC-BY-3.0 licenses](https://creativecommons.org/licenses/by/3.0/).
According with [this post](https://creativecommons.org/version4/) the [CC-BY-4.0 licenses](https://creativecommons.org/licenses/by/4.0/) are more user-friendly and more internationally robust than ever before. [Extra reading](https://creativecommons.org/faq/#what-are-the-international-unported-creative-commons-licenses-and-why-does-cc-offer-ported-licenses)

In FreeCAD 0.19, Link branch and 0.20 versions in the `Project information` you can not choose the CC-BY-3.0 license because the suite of licenses is the 4.0 one, so, in order to have coherence between the FreeCAD software and the library a change is needed.

Is it viable to just change the README.md file or not?

Thorbijoern, 2022-06-09:
Sorry that i didn't reply to this Issue sooner. I must have missed the section in the Readme, sorry.
I don't know anymore what my initial issue was, maybe it was late. but i think the initial issue is solved.
I'm not an expert with licenses but the CC-BY 4.0 seems to have some great changes.
I also don't know how a licence change works on a project, but i think there are good reasons to update the licence as @hasecilu stated.
I can't really comment on the matter if other licences would be better...
I only know that libraries or frameworks for writing software are generally licenced more open, like with the MIT licence, but a complete programm uses a stricter licence like GPL. I don't know how that compares to CC but the CERN OHL also has similar variants: https://software.development.fabcity.hamburg/osh-license-discussion/01-introduction/