FreeCAD/FreeCAD-library #335: library dimension: something wrong?

itarozzi, 2021-11-21:
I'm trying to download the library as zip or using github clone.

The zip is about 4.4GB but decompressed is more over. Can't be more precise because it full my disk, but Mechanical Parts/Chains are about 15GB

In the readme i can read a dimensione of about 1GB. What happens?

Is there a way to download only a sub-part of the library?

pysimone, 2021-12-07:
You can try with "sparse-checkout"

For example if you want to download only "Electrical Parts/Batteries" folder:

`git clone --filter=blob:none --no-checkout https://github.com/FreeCAD/FreeCAD-library.git`
`cd FreeCAD-library`
`git sparse-checkout set 'Electrical Parts'/Batteries`

itarozzi, 2021-12-09:
Ok, this could be a workaround for experienced users, I'll try it.

But for installing the library using the Addon Manager is not really working. 

But my question is: why the library is so huge? > 18GB ?!?

luzpaz, 2021-12-10:
@chennes I heard you're looking at the Addon Manager. Do you have the bandwidth to put this on your todo list?  
CC @yorikvanhavre

luzpaz, 2021-12-10:
@chennes so most likely
1. Going in and optimizing models
2. Breaking the Library up in to submodules ?

chennes, 2021-12-13:
Some discussion here: https://forum.freecadweb.org/viewtopic.php?t=63140