FreeCAD/FreeCAD-library #341: Chain Sprockets take up a lot of disk space

alexneufeld, 2022-01-01:
cc @pysimone

I cloned this repo for the first time in a while... and filled my hard drive.

Let's inspect the disk usage:
|path|size|
|---|---|
|root|33.0 GB|
|/.git|5.0 GB|
|/Mechanical Parts/Chains/Sprocket/ISO 606|26.0 GB|

so one class of parts takes up >75% of the space,  mostly occupied by stl files.

I don't have an easy solution to this problem, but I do think that there needs to be some optimization done to the folder in question. 

At the very least, the README should be updated to give a more accurate description of the downloaded size. (maybe try to update it monthly with a +/-10% figure?)

luzpaz, 2022-01-02:
@Obijuan ping

pysimone, 2022-01-02:
I tried to optimize the size of the parts, but I didn't get anything good.
I stop uploading files, to not to take up too much space, until a solution is found.

Obijuan, 2022-01-02:
I will stop merging new parts until a solution is find

alexneufeld, 2022-01-02:
some preliminary suggestions:

1. store an stl preview for only a small sample of the parts. Since most of the sprockets  are topologically identical except for the pitches and tooth counts, the user can infer what most of them look like.

E.G:
delete all existing stl previews, and replace them with:
/Mechanical Parts/Chains/Sprocket/ISO 606/triplex_example.stl
/Mechanical Parts/Chains/Sprocket/ISO 606/duplex_example.stl
/Mechanical Parts/Chains/Sprocket/ISO 606/simplex_example.stl

2. Make sure stl files are stored in binary format. should give a significant space reduction compared to ASCII

make sure to keep an eye on the size of the .git directory. Git uses aggressive compression to keep it small, but we do need to make sure that anything we do doesn't blow *that* set of files up to some unmanageable size...

pysimone, 2022-01-03:
All right.
I will recursively delete all existing stl files under the "chains" folder and I will put only example stl files.
Then, I will not upload anymore stl files, except the example ones.

alexneufeld, 2022-01-04:
#342 
Thanks everyone for the quick turnaround.

closed 2022-01-04 by alexneufeld