FreeCAD/FreeCAD-library #357: directions for linking to other repositories

kinghat, 2022-04-09:
looking for the best way to go about linking parts from a current repository that my projects live in, to a fork of the FreeCAD-library, so people can pr from there. the most simple way is to manually copy/paste the parts but the former should be possible with submodules or file linking.

luzpaz, 2022-04-09:
Interesting. Please discuss on forum as this issue queue gets very little love. At some point, and this is IMHO, we should have an asset manager that leverages the Library and the future Assembly workbench.

hasecilu, 2022-04-17:
Do you pretend to maintain specific models or some folder?
Last months I have been contributing some models that I needed for an specific project. I don't know how to manage submodules or that juicy git stuff.

I just downloaded all library, make my changes and then submit here.
To use in my project I create a folder called `cad/FreeCAD-library` and there I copy the files I need, some that I created and some from the community.
The main reason that I just copied them is because I need different sizes of same model or the model is parametric (spreadsheet) or I made small changes to match the real components that I already have and also add `LCS` to use with `Assembly4` and `Assembly4`modify the models so `git` will detect a change.

If you use Linux you can use `Symbolic links` (ln -s path/to/file) so you can modify once and have the changes everywhere.