FreeCAD/FreeCAD-library #382: freecad library preview show only some model

sdifru, 2023-01-24:
I opened **mechanical parts/bearings**, clicked **608ZZ_Ball_Bearing.fcstd**, and it showed a preview. Then I clicked **623ZZ_Ball_Bearing.fcstd** and it didn't show any preview. I've only tested it on the bearings library. Some libraries show previews and others don't.

What could be wrong? Are there things I should tweaks?

chennes, 2023-01-24:
In GitHub only STL files will actually show a 3D preview: no FCStd files should do so. Can you confirm that you saw a preview of a file when specifically choosing the FCStd file, and not its STL version?

sdifru, 2023-01-24:
> In GitHub only STL files will actually show a 3D preview: no FCStd files should do so.

I didn't see the preview in github. I saw it in FreeCAD app.

> Can you confirm that you saw a preview of a file when specifically choosing the FCStd file, and not its STL version?

Yes. I clicked the FCStd files. The STL ones don't show previews.

When we run the `PartsLibrary.FCMacro`, it shows part libraries. I clicked the files there to see previews. Some FCStd files showed preview and some don't.

This one show a preview:
[image]

This one doesn't:
[image]

chennes, 2023-01-24:
Oh, I see. Yes, some of the FCStd files in this library have embedded preview images and some don't. It would be a good project for someone (doesn't have to be a developer) to download those files and resave them with previews, then submit a PR.

sdifru, 2023-01-24:
So it happened because some files doesn't have preview images embedded on them. I thought I did something wrong with the installation. I'll close this issue. Thanks @chennes !

closed 2023-01-24 by sdifru