FreeCAD/FreeCAD-library #391: `SMLogo.SVG` popup message at start

luzpaz, 2023-05-31:
[image]

forum thread: https://forum.freecad.org/viewtopic.php?t=78649

```
OS: macOS 13.3
Word size of FreeCAD: 64-bit
Version: 0.21.0.31483 (Git)
Build type: Release
Branch: master
Hash: 588620fd9fd138892b1cd2466f5c87d441493174
Python 3.11.0, Qt 5.15.6, Coin 4.0.0, Vtk 9.2.2, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.0.0
  * parts_library 0.2.63
  * A2plus 0.4.60n
  * BIM 2021.12.0
  * fasteners 0.4.56
  * Assembly4 0.50.2
  * Assembly3 0.12.0
```

luzpaz, 2023-06-10:
@Obijuan @yorikvanhavre can you recreate ?

We made need a macOS user to try to recreate

yorikvanhavre, 2023-06-12:
I don't understand where this comes from... Trying to pull the parts library now to see...

yorikvanhavre, 2023-06-12:
Just tested, and I don't see the message