FreeCAD/FreeCAD-library #407: parts library doesnt work for me under windows

urishab, 2024-01-28:


shaise, 2024-01-28:
try C:\Users%username%\AppData\Roaming\FreeCAD\Mod\FreeCAD_library

urishab, 2024-01-28:
I think I solved this by restarting FreeCAD
I also submitted a PR to add some instructions on using this addon as it was very unclear to me

closed 2024-01-28 by urishab

shaise, 2024-01-28:
Yes, they have installation instructions but not usage.
Also I do not think the plugin should hold the entire library locally.

urishab, 2024-01-28:
They also write in the instructions that the BIM addon has a module for using the parts library in online-mode. I've tried it but it seems to be broken.

shaise, 2024-01-28:
I'm working on the BIM add-on. I will try to revive the online-mode. If it works, perhaps I will create a separate macro for it, as it is useless to hold 5GB of parts locally.

shaise, 2024-01-28:
I have fixed the broken bim on line library browser. I placed a pr for it.
Meanwhile if you want to test: https://github.com/shaise/BIM_Workbench