FreeCAD/FreeCAD-library #411: Part Library browser cannot access hidden directory below "~/.local"

wolfgangr, 2024-02-29:
- installed part library using addon manager
- called macro as instructed
- can browse file tree, but not access part library location
- opening part library in addon manager, I find the installation path:
`/home/username/.local/share/FreeCAD/Mod/parts_library`
but cannot open this location, since **directory names starting with a dot** are **hidden** in standard GUI
- don't know how the file browser of the GUI is configured, but I can't find a location to enter/show hidden directories
- **workaround:** create a non hidden symbolic link, e.g.
`_local_share_FreeCAD -> /home/username/.local/share/FreeCAD`

system details:
```
OS: Debian GNU/Linux 10 (buster) (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.5.12
  * slic3r-tools
  * Pyramids-and-Polyhedrons
  * A2plus 0.4.63
  * Assembly3 0.12.2
  * Assembly4 0.50.7
  * FeedsAndSpeeds 0.5.0
  * dodo 1.0.0
  * frame 0.1.1
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * MOOC 2022.4.21
  * Help 1.0.3
  * ose-piping
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.2
  * ThreadProfile 1.89.0
  * SteelColumn
  * Estimate 0.1.2
  * DynamicData 2.60.0
  * parts_library
```

k2-mirror, 2024-08-19:
The problem is that for some reason the PartsLibrary.FCMacro script is not copied when installing the Part Library. I have solved it by downloading PartsLibrary.FCMacro file from github, and copying it to /home/username/.local/share/FreeCAD/Macro

Toutoune86, 2024-11-17:
Is there some nice solution for the same issue in windows ?

galou, 2025-09-22:
Did you try right-clicking somewhere in the list of directories/files in the "Open file" dialog? On my system (Ubuntu with FreeCAD from PPA), this proposes the option to show hidden files.