FreeCAD/FreeCAD-library #434: Break up library into smaller libraries

davija, 2024-12-17:
At this point, this library is HUGE as indicated elsewhere and in documentation.  This library contains a LOT of things that most people won't use.  This library should be broken up into a collection of smaller libraries to reduce the overall size of each library.  I guess it makes sense that the large library could still be kept as-is, but it would be far better for users to be able to select just the components that re useful to them.

yorikvanhavre, 2024-12-18:
That's an excellent idea. Would you make a proposal on how to split up?

Obijuan, 2024-12-18:
Great! I fully agree. In the process of breaking up the library we can also update some components to Freecad 1.0. I am not sure how to split it up, but al least there should be a collection called something like "electronic componets". In there we can put all the 3D models of the components used in PCBs (very useful for using with tools like Kicad). I can be in charge of updating this collection

chennes, 2024-12-19:
This is basically superseded by the work done in https://github.com/FreeCAD/FreeCAD-Component-Management-System -- we're basically just waiting on @kkremitzki to help us getting the server set up, then it's ready for public use.

luzfcb, 2025-04-19:
It is possible to reduce the download size a bit by limiting the clone depth, by adding the `--depth 1` on the git clone command. Today, this resulted in 4.35GB of download.

On OSX is something like: `git clone --depth 1 --recurse-submodules https://github.com/FreeCAD/FreeCAD-library "/Users/<username>/Library/Application Support/FreeCAD/Mod/parts_library"`

Additionally, would probably use it too ` --single-branch --jobs 4`

Mambix, 2025-09-09:
> That's an excellent idea. Would you make a proposal on how to split up?

IMO it should be done like FreeCAD-addons is setup, a collection of submodules. And each folder in this repo would be its separate library. 

But what ever solution you do it needs to solve one big issue for users on slower connections. You need to be able to fetch data even if you're not on fiber. Even here in Asia I have fiber internet in the office and I can not pull the repo out even using `--depth 1` trick :)

p.s. If I would have a magic wand to design this I would make it as a [Resilio](https://www.resilio.com/) storage. Maintainers get write access to update models. Every one else that wants access gets a readonly key. And then everyone of the users can host data for other users; yes it is a torrent protocol based system. So it doesnt matter where you access it from the system will find fastest points of download for anyone :) But I don't think Resilio has any free plans for sharing opensource data.