FreeCAD/FreeCAD-library #441: FreeCAD crashes when deleting kitchen cabinet

j-hap, 2025-04-30:
Steps to reproduce:
- Open FreeCAD 1.0.0
- Switch to BIM
- Create a new Document
- Open Object Library Browser
- Insert vertical kitchen cabinet
- Delete the just inserted cabinet
- FreeCAD crashes

Demo:
[image]

It probably is a bug in FreeCAD itself, but it is triggered by the component in the library, so I'm not sure if the issue should be here or in the FreeCAD project.

Version info
```
OS: Fedora Linux 42 (Workstation Edition) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.1, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * DynamicData 2.71.0
  * fasteners 0.5.37
```

Crash dump:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x19c30) [0x7f0ead645c30]
#1  0x7f0eb0ddf1cc in Gui::DocumentObjectItem::testStatus(bool, QIcon&, QIcon&) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0x41c
#2  0x7f0eb0de0992 in Gui::DocumentItem::testStatus() from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0x72
#3  0x7f0eb0e0c10a in Gui::TreeWidget::onUpdateStatus() from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0x93a
#4  0x7f0eb0e0febb in Gui::TreeWidget::onItemSelectionChanged() from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0x18b
#5  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7f0eadd1684a]
#6  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Widgets.so.6(+0x1bbb8b) [0x7f0eaedbbb8b]
#7  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7f0eadd1684a]
#8  0x7f0eadb51045 in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6+0x45
#9  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6(+0x1ea6ad) [0x7f0eadbea6ad]
#10  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7f0eadd1684a]
#11  0x7f0eadb50923 in QAbstractItemModel::rowsAboutToBeInserted(QModelIndex const&, int, int, QAbstractItemModel::QPrivateSignal) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6+0x63
#12  0x7f0eadbf5764 in QAbstractItemModel::beginInsertRows(QModelIndex const&, int, int) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6+0x74
#13  0x7f0eaee466b0 in QTreeWidgetItem::insertChild(int, QTreeWidgetItem*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Widgets.so.6+0xf0
#14  0x7f0eb0e06abf in Gui::DocumentItem::createNewItem(Gui::ViewProviderDocumentObject const&, QTreeWidgetItem*, int, std::shared_ptr<Gui::DocumentObjectData>) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0xff
#15  0x7f0eb0e12922 in Gui::TreeWidget::_slotDeleteObject(Gui::ViewProviderDocumentObject const&, Gui::DocumentItem*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0x3d2
#16  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so(+0x4e317f) [0x7f0eb0ae317f]
#17  0x7f0eb0b4c743 in Gui::Document::slotDeletedObject(App::DocumentObject const&) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0x153
#18  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADApp.so(+0x23a71f) [0x7f0eafa3a71f]
#19  0x7f0eafa56cdb in App::Document::removeObject(char const*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADApp.so+0xfb
#20  0x7f0eafa993d4 in App::GroupExtension::removeObjectFromDocument(App::DocumentObject*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADApp.so+0xc4
#21  0x7f0eafa9943c in App::GroupExtension::removeObjectsFromDocument() from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADApp.so+0x2c
#22  0x7f0eafa9f2d3 in App::GroupExtensionPy::removeObjectsFromDocument(_object*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADApp.so+0x33
#23  0x7f0eafa9f314 in App::GroupExtensionPy::staticCallback_removeObjectsFromDocument(_object*, _object*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADApp.so+0x24
#24  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libpython3.13.so.1.0(+0x1daed3) [0x7f0eb01daed3]
#25  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libpython3.13.so.1.0(_PyObject_MakeTpCall+0x2c4) [0x7f0eb01b74d4]
#26  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libpython3.13.so.1.0(+0x9fe79) [0x7f0eb009fe79]
#27  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libpython3.13.so.1.0(PyEval_EvalCode+0xad) [0x7f0eb027af1d]
#28  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libpython3.13.so.1.0(+0x2b7ced) [0x7f0eb02b7ced]
#29  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libpython3.13.so.1.0(+0x2b5602) [0x7f0eb02b5602]
#30  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libpython3.13.so.1.0(+0x2b135e) [0x7f0eb02b135e]
#31  0x7f0eaf54651c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADBase.so+0x6c
#32  0x7f0eb0be33f4 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0x94
#33  0x7f0eb0be357f in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0xdf
#34  0x7f0eb0fe0635 in Gui::ViewProviderGroupExtension::extensionOnDelete(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0x205
#35  0x7f0eb0fd7cca in Gui::ViewProvider::onDelete(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0x10a
#36  0x7f0eb0c04abd in StdCmdDelete::activated(int) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0x63d
[... 12 lines cut ...]
#49  0x7f0eae7e0ff6 in QShortcutMap::dispatchEvent(QKeyEvent*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Gui.so.6+0x446
#50  0x7f0eae7e1a61 in QShortcutMap::tryShortcut(QKeyEvent*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Gui.so.6+0x81
#51  0x7f0eae43b939 in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Gui.so.6+0x199
#52  0x7f0eae3cda8f in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Gui.so.6+0x1bf
#53  /home/j-hap/miniforge3/envs/freecad/lib/qt6/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so(+0x12b8a) [0x7f0e99ee7b8a]
#54  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6(+0x31684a) [0x7f0eadd1684a]
#55  0x7f0ead591b64 in QDBusPendingCallWatcher::finished(QDBusPendingCallWatcher*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/./libQt6DBus.so.6+0x44
#56  0x7f0eadce30ca in QObject::event(QEvent*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6+0x24a
#57  0x7f0eaf1b3e48 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Widgets.so.6+0x88
#58  0x7f0eb0b91389 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0xd9
#59  0x7f0eadd5fb8a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6+0x17a
#60  0x7f0eadd8e2ae in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6+0x3ce
#61  /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6(+0x1019b8) [0x7f0eadb019b8]
#62  /home/j-hap/miniforge3/envs/freecad/bin/../lib/./libglib-2.0.so.0(+0x5b5e6) [0x7f0eac5085e6]
#63  /home/j-hap/miniforge3/envs/freecad/bin/../lib/./libglib-2.0.so.0(+0x5e7b7) [0x7f0eac50b7b7]
[... cut ...]
#66  0x7f0eadd9747c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6+0x25c
#67  0x7f0eadd977da in QCoreApplication::exec() from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libQt6Core.so.6+0x9a
#68  0x7f0eb0ae93a5 in Gui::Application::runApplication() from /home/j-hap/miniforge3/envs/freecad/bin/../lib/libFreeCADGui.so+0x8a5
#69  freecad(+0x7b67) [0x55a875083b67]
#70  /lib64/libc.so.6(+0x35f5) [0x7f0ead62f5f5]
#71  /lib64/libc.so.6(__libc_start_main+0x88) [0x7f0ead62f6a8]
#72  freecad(+0x7eda) [0x55a875083eda]
```

bitbasher, 2025-05-06:
i was able to add a vertical cabinet from the library in a dev version of FC that I am playing with right now.  I saw a lot of interesting warnings in the report panel as it was added .. this is the [pastebin link to see them ](https://bpa.st/EXAQ) all.

a possible factor .. i had a message about a git related "thing" not being installed,
here are the first few notifications  :
```
12:25:26  BIM workbench activated
12:26:08  Unit system updated for active document
12:27:08  python-git not found. Git-related functions will be disabled
12:27:49  Loading Spreadsheet module... done
12:27:49  Loading GUI of Spreadsheet module... done
12:27:49  <ElementMap> C:/Users/Jeff/AppData/Roaming/FreeCAD/Macro/PartsLibrary.FCMacro(266): Invalid element name string id
12:27:49  <ElementMap> C:/Users/Jeff/AppData/Roaming/FreeCAD/Macro/PartsLibrary.FCMacro(266): Missing element name prefix id
```
--
then came several hundred more warnings that you can see in the pastebin file.

I open the vertical cabinet file in my dev version of FreeCAD into an empty document .. NOT in the BIM workbench, and i get all the same warnings. I save it into a new document elsewhere. I rename the existing vertical cabinet file to hide it, then copy in my just saved file.
Now I start a new BIM document, so i am in the BIM workbench, open the library selector and bring in my updated vertical cabinet file .. no warning messages. I try to delete it and get a TON of errors and warnings and only parts of it actually delete .. it leaves a horrible mess in the Model Tree.
The original file is still visible in the Kitchen directory so i try to bring it in again .. and boom .. all of the original problems show.

So it looks like the .FCStd files in the Library will all have to be updated. Ouch

-- my FreeCAD version info --
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.41264 (Git) Conda
Build date: 2025/04/14 16:28:55
Build type: Release
Branch: main
Hash: 31b0dbaa79da9e0a8d18c009006a2b1c6cd16404
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/Belgium (en_BE) ]
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * 3D_Printing_Tools
 ... blah blah
  * parts_library
 ... more blah

semhustej, 2026-01-20:
The cabinet model is a complex Part Design model. I can reproduce getting lot of errors when trying to delete the parent group. To remove the model completely, you have to than also remove all the remaining parts, which are left after the group is deleted.

```
OS: Debian GNU/Linux 13 (trixie) (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.1.0rc2.20260108 (Git shallow) AppImage
Build date: 2026/01/08 12:03:27
Build type: Release
Branch: (HEAD detached at 0c7935c)
Hash: 0c7935c755bd9531f7061be31fab8788742a44b7
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.4, OCC 7.8.1
Locale: English/Ireland (en_IE)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1

```