FreeCAD/FreeCAD-library #447: Assembly workbench Transform. Link problem

ashnansen, 2025-09-20:
When using part in Assembly mode the default action of Transform.  Should of  not also move the part. In the file that the part  from. 
In the picturs the part is the pink part and the food print of an sketch of the part is the whit lines on the Origin planes. 

If the part are being used in other assemblies there an problem. With alignment in projects.   

I am using FreeCAD 1.0.2

[image]
[image]
[image]
[image]

luzpaz, 2025-09-20:
Is this a ticket referring to the FreeCAD-Library repository or FreeCAD itself ? If the latter then you're posting to the wrong repo, it should be https://github.com/FreeCAD/FreeCAD

ashnansen, 2025-09-21:
I can see that I have posted the last to posts in FreeCAD-Libarry

luzpaz, 2025-09-22:
@ashnansen that didn't address the question i was asking. 

note, you may want to test FreeCAD 1.1.0dev since assembly improved immensly in this upcoming version. Please use that version to report issues with the Assembly wb as I don't think 1.0.x will see more significant assembly improvements. 

I'm closing this.